alexa-pi/AlexaPi #404: Wake word works but poor audio

TCPMeta, 2018-06-19:
Using a RaspberryPi 3 Plus with Raspian Stretch.
Using the audio jack instead of HDMI for audio output and using a USB Mic for audio Input.

With the defaults I can say "Alexa" and she says "Yes?" then I issue a command nothing happens or I'll get pop sounds. I tried using Pulse and even followed the guide for a system wide Pulse setup and it just doesn't like it.

I haven't tried using SoX yet for the handler. From reading a bunch of the choppy sound issues it seems to be something wrong with the latest version of AlexaPi, Pulse or Raspian.

Anyone found the correct mixture of the three that works?

nicusorflorin, 2018-06-30:
i got the same issue, i think it's raspian's fault, anyone any clue?

EmerickH, 2018-07-19:
Duplicate of #239

Hello,
VLC is not working very well in Raspbian, this is an already known bug.
You must use SoX.
Here is a tutorial to setup PulseAudio in AlexaPi with SoX:
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended

closed 2018-07-19 by EmerickH