alexa-pi/AlexaPi #406: Headless PiZero with HifiberryMiniamp and Squeezlite

dersch81, 2018-07-07:
Hi, I have an output of alexaPi as follows:

`Jul 07 21:43:26 KloPi python[370]: INFO: Checking Internet Connection ...
Jul 07 21:43:27 KloPi python[370]: INFO: Connection OK
Jul 07 21:43:27 KloPi python[370]: INFO: AVS token: Requesting a new one
Jul 07 21:43:28 KloPi python[370]: INFO: AVS token: Obtained successfully
Jul 07 21:43:29 KloPi python[370]: [01e87270] alsa audio output error: cannot open ALSA device "default": Device or resource busy
Jul 07 21:43:29 KloPi python[370]: [01e87270] core audio output error: Audio output failed
Jul 07 21:43:29 KloPi python[370]: [01e87270] core audio output error: The audio device "default" could not be used:
Jul 07 21:43:29 KloPi python[370]: Device or resource busy.
Jul 07 21:43:29 KloPi python[370]: [01e87270] core audio output error: module not functional
Jul 07 21:43:29 KloPi python[370]: [b3545020] core decoder error: failed to create audio output`

As is said i'm trying to run it on an headless PiZero without desktop and an Hifiberry MiniAmp and using it with Squeezlite. The Hifiberry Amp is the default audio output. Does squeezelite affect the audio output of AlexaPi? I'm unsure at the moment if i have to follow the Pulseaudio steps because i don't use a desktop installation.

EmerickH, 2018-07-19:
Hello,
Alsa can't handle multiples applications using a device at the same time.
Try to :
 * Use `dmix` PCM:
   ```
   output_device: "dmix"
   ```
   in */etc/opt/AlexaPi/config.yaml*
   (You must also change device in Squeezlite)
 * [Use PulseAudio](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended)

dersch81, 2018-07-19:
Ok thanks. now i can start AlexaPi while squeezelite is running. But it is still not working. I'm using a Ps3Eye and have some troubles. I just hear the "hello" after AlexaPi is started. I found that within the logs: 

`Jul 19 23:35:45 KloPi python[373]:     self.run()
Jul 19 23:35:45 KloPi python[373]:   File "/usr/lib/python2.7/threading.py", line 754, in run
Jul 19 23:35:45 KloPi python[373]:     self.__target(*self.__args, **self.__kwargs)
Jul 19 23:35:45 KloPi python[373]:   File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 77, in thread
Jul 19 23:35:45 KloPi python[373]:     data = self._capture.handle_read()
Jul 19 23:35:45 KloPi python[373]:   File "/opt/AlexaPi/src/alexapi/capture.py", line 119, in handle_read
Jul 19 23:35:45 KloPi python[373]:     return self._handle.read(self._handle_chunk_size, exception_on_overflow=self._pa_exception_on_overflow)
Jul 19 23:35:45 KloPi python[373]:   File "/usr/local/lib/python2.7/dist-packages/pyaudio.py", line 608, in read
Jul 19 23:35:45 KloPi python[373]:     return pa.read_stream(self._stream, num_frames, exception_on_overflow)
Jul 19 23:35:45 KloPi python[373]: IOError: [Errno -9999] Unanticipated host error
`

That's my audio setting for AlexaPi:

`sound:
  # Name of your microphone device: "raspberrypi"
  # leave empty for default
  input_device: "USB Camera-B4.09.24.1: Audio (plughw:1,0)"

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "dmix:CARD=sndrpihifiberry,DEV=0"
`

The Input device i choosed because of the log entries. With some other device ex. sysdefault:CARDxxxxx or other i couldn't start alexaPi.

EmerickH, 2018-07-20:
Seems to be a PyAudio bug, try to update it:
```
sudo pip install pyaudio 
```
If it's not working, check the version with:
```
pip show pyaudio 
```
source: https://stackoverflow.com/questions/44351501/exception-on-overflow-parameter-not-working-pyaudio

EmerickH, 2018-07-20:
I think `dmix` is using rates that are not supported by PyAudio, try to [setup PulseAudio](https://emerickh.github.io/AlexaPi-config/setup/pulse.html) instead with SoX if you are on Raspberry Pi or some ARM devices.

dersch81, 2018-07-20:
Ok I can try. It is a little PiZero don't you think PulseAudio is a bit too much for that device?

EmerickH, 2018-07-20:
I'm running PA on an old Raspberry Pi B that is less performant than a Pi Zero without any problem so I think you'll not have any problem.

ErfolgreichCharismatisch, 2019-09-19:
> I'm running PA on an old Raspberry Pi B that is less performant than a Pi Zero without any problem so I think you'll not have any problem.

I am trying to install on jessie on a Raspberry Pi B. It cannot find vlc-bin. What to do?