alexa-pi/AlexaPi #407: Audio Setup problem for VLC and SoX - need advice

shinyone, 2018-07-11:
**Your OS (including version) where you are running AlexaPi:**
```
Rapsbian
```

**Your hardware platform and model you are running on:**
```
Rasp pi 3+
```

**Python release (`python2 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
I have a problem with audio setup and don't know how to fix it.   
I have tried both vlc and sox and followed the instructions for [PulseAudio](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging).

With vlc:

[6c6004e0] pulse audio output error: PulseAudio server connection failure: Access denied
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

[6c6004e0] alsa audio output error: cannot open ALSA device "default": Connection refused
[6c6004e0] core audio output error: Audio output failed
[6c6004e0] core audio output error: The audio device "default" could not be used:
Connection refused.
[6c6004e0] core audio output error: module not functional
[73837a78] core decoder error: failed to create audio output

With sox:

 DEBUG: Error attempting play. sox FAIL formats: can't open output file `default': can not open audio device: Access denied

Same with pulseaudio / alsa
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```
sox
  input_device: "USB PnP Sound Device: Audio (plughw:1,0)"
  output: "pulseaudio"
  output_device: ""
  playback_handler: "sox"

vlc
  input_device: "USB PnP Sound Device: Audio (plughw:1,0)"
  output: "pulse"
  output_device: ""
  playback_handler: "vlc"

```

**Steps to reproduce:**

1. Clean install
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
 http://www.alsa-project.org/db/?f=d9b972e381a285c30c5226566684278ed89b6da8
```

shinyone, 2018-07-11:
I saw this in the log

> 2018-07-11 16:42:11 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2018-07-11 16:42:11 DEBUG: Started play. sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio vol -6 dB pad 0 0
2018-07-11 16:42:11 DEBUG: Indicate Start Playing
2018-07-11 16:42:11 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2018-07-11 16:42:11 DEBUG: Response: {"status":"success"}
2018-07-11 16:42:11 DEBUG: Error attempting play. sox FAIL formats: can't open output file `default': can not open audio device: Access denied

I am able to run this directly from the prompt

`sox -q /opt/AlexaPi/src/resources/hello.mp3 -t pulseaudio vol -6 dB pad 0 0`

EmerickH, 2018-07-19:
Seems to be a duplicate of never solved #255
Are you starting alexaPi manually or with the service?
Try:
```
sudo adduser alexapi pulse-access
sudo adduser alexapi audio
```

shinyone, 2018-07-19:
Starting with service.

Thanks for the advice, i tried both, but without success.:

```
pi@magicpi:~ $ sudo adduser alexapi pulse-access
The user `alexapi' is already a member of `pulse-access'.
pi@magicpi:~ $ sudo adduser alexapi audio
The user `alexapi' is already a member of `audio'.
```

And after changing the config

> 2018-07-19 14:37:16 CRITICAL: Your input_device 'pulse' is invalid. Use one of the following:
> USB PnP Sound Device: Audio (plughw:1,0)

> 2018-07-19 14:34:58 CRITICAL: Your input_device 'plughw:1,0' is invalid. Use one                                                                                              of the following:
> USB PnP Sound Device: Audio (plughw:1,0)

shinyone, 2018-07-19:
Running manually to get more error information via:
`sudo /opt/AlexaPi/src/main.py -d`

Not having luck with vlc or sox.

```
INFO: feat.c(715): Initializing feature stream to type: '1s_c_d_dd', ceplen=13, CMN='batch', VARNORM='no', AGC='none'
INFO: acmod.c(162): Using subvector specification 0-12/13-25/26-38
INFO: mdef.c(518): Reading model definition: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/mdef
INFO: mdef.c(531): Found byte-order mark BMDF, assuming this is a binary mdef file
INFO: bin_mdef.c(336): Reading binary model definition: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/mdef
INFO: bin_mdef.c(516): 42 CI-phone, 137053 CD-phone, 3 emitstate/phone, 126 CI-sen, 5126 Sen, 29324 Sen-Seq
INFO: tmat.c(149): Reading HMM transition probability matrices: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/transition_matrices
INFO: acmod.c(113): Attempting to use PTM computation module
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/means
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size:
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/variances
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size:
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(304): 222 variance values floored
INFO: ptm_mgau.c(475): Loading senones from dump file /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/sendump
INFO: ptm_mgau.c(499): BEGIN FILE FORMAT DESCRIPTION
INFO: ptm_mgau.c(562): Rows: 128, Columns: 5126
INFO: ptm_mgau.c(594): Using memory-mapped I/O for senones
INFO: ptm_mgau.c(837): Maximum top-N: 4
INFO: phone_loop_search.c(114): State beam -225 Phone exit beam -225 Insertion penalty 0
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2018-07-19 14:42:20 DEBUG: Setting up playback handler: VlcHandler
2018-07-19 14:42:20 DEBUG: Setting up Magic Mirror platform
2018-07-19 14:42:20 INFO: Magic Mirror HTTP Server - :5055
2018-07-19 14:42:20 INFO: Checking Internet Connection ...
2018-07-19 14:42:20 DEBUG: Starting new HTTPS connection (1): api.amazon.com:443
2018-07-19 14:42:21 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2018-07-19 14:42:21 INFO: Connection OK
2018-07-19 14:42:21 INFO: AVS token: Requesting a new one
2018-07-19 14:42:21 DEBUG: Starting new HTTPS connection (1): api.amazon.com:443
2018-07-19 14:42:22 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2018-07-19 14:42:22 INFO: AVS token: Obtained successfully
2018-07-19 14:42:22 DEBUG: Starting Magic Mirror platform HTTP Server
2018-07-19 14:42:22 DEBUG: Starting Magic Mirror heartbeat with 5 second interval
2018-07-19 14:42:22 DEBUG: Sending MM Heatbeat
2018-07-19 14:42:22 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:22 DEBUG: Stopping audio play
2018-07-19 14:42:22 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2018-07-19 14:42:22 DEBUG: Player State: State.Opening
2018-07-19 14:42:22 DEBUG: Player State: State.Playing
2018-07-19 14:42:22 DEBUG: Started play.
2018-07-19 14:42:22 DEBUG: Indicate Start Playing
2018-07-19 14:42:22 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2018-07-19 14:42:22 DEBUG: Response: {"status":"success"}
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

[6c6004e0] alsa audio output error: cannot open ALSA device "default": Connection refused
[6c6004e0] core audio output error: Audio output failed
[6c6004e0] core audio output error: The audio device "default" could not be used:
Connection refused.
[6c6004e0] core audio output error: module not functional
[72a35f68] core decoder error: failed to create audio output
2018-07-19 14:42:24 DEBUG: Player State: State.Ended
2018-07-19 14:42:24 DEBUG: Finished play.
2018-07-19 14:42:24 DEBUG: Indicate Stop Playing
2018-07-19 14:42:24 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2018-07-19 14:42:24 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:24 DEBUG: Indicating Success
2018-07-19 14:42:24 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
2018-07-19 14:42:24 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:27 DEBUG: Sending MM Heatbeat
2018-07-19 14:42:27 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:32 DEBUG: Sending MM Heatbeat
2018-07-19 14:42:32 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:37 DEBUG: Sending MM Heatbeat
2018-07-19 14:42:37 DEBUG: Response: {"status":"success"}
2018-07-19 14:42:42 DEBUG: Sending MM Heatbeat
2018-07-19 14:42:42 DEBUG: Response: {"status":"success"}
^C2018-07-19 14:42:42 DEBUG: Cleaning up playback handler
2018-07-19 14:42:42 DEBUG: Cleaning up Magic Mirror platform
```

shinyone, 2018-07-19:
Problem seems more on output than input.

```
pi@magicpi:~ $ aplay -L
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```

EmerickH, 2018-07-19:
Very strange, I've never seen that...
Try to reinstall PulseAudio:
```
sudo apt-get remove --purge pulseaudio
sudo apt-get autoremove
sudo apt-get install pulseaudio
```

shinyone, 2018-07-19:
no same problem.

closed 2018-07-19 by shinyone

EmerickH, 2018-07-19:
Ok, good to know!