alexa-pi/AlexaPi #409: How to force Alexa to talk to me in english ?

zdaar, 2018-07-22:
Hello, 

I setup an AlexaPI and it is working fine.
However, she talks back to me in french and I do not want that. I'd like her to speak english.
How to force it ?

Thanks in advance

EmerickH, 2018-07-23:
Très étrange! Il semblerait que Amazon ait intégré les autres langues à l'ancienne API? (normalement pas possible #328). Essayez de changer la langue dans les paramètres sur https://alexa.amazon.fr/

Very strange! It seems that Amazon has integrated other languages to the old API? (normally not possible #328). Try to change the language in the settings on https://alexa.amazon.fr/

EmerickH, 2019-01-07:
@zdaar You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH