alexa-pi/AlexaPi #411: Problem with German language

Uenne, 2018-08-10:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

Uenne, 2018-08-10:
I'm from Germany, so excuse my mistakes.
I've installed AlexaPi on Raspberry Model 3B+ and Stretch.
Installation etc went without problems, but I have aproblem with the language.
Starting Alexa, it answered with "Yes" and not "Ja" (German), but this ic a minor point. When I speak in german, Alexa is responding in German. But I can't activate German skills. The exact location is given in the Alexa App. For the daily summary (I hope it is the correct translation) I can select between Fox, CNN and Reuters, no german News.
It seems to me as if it a mixture of US and German.

Any idea?
Thanks for your support.
Jürgen

closed 2018-08-10 by Uenne

elka0711, 2018-12-30:
I have exactly the same problem