alexa-pi/AlexaPi #412: alexapi

tinkerermaruthu, 2018-08-12:
pi@raspberrypi:~ $ sudo systemctl start AlexaPi.service
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor pres
   Active: active (running) since Sun 2018-08-12 03:38:33 UTC; 1h 4min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 303 (python)
   CGroup: /system.slice/AlexaPi.service
           └─303 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Aug 12 03:39:47 raspberrypi python[303]:     self.__target(*self.__args, **self.
Aug 12 03:39:47 raspberrypi python[303]:   File "/opt/AlexaPi/src/alexapi/trigge
Aug 12 03:39:47 raspberrypi python[303]:     self._capture.handle_init(self.AUDI
Aug 12 03:39:47 raspberrypi python[303]:   File "/opt/AlexaPi/src/alexapi/captur
Aug 12 03:39:47 raspberrypi python[303]:     frames_per_buffer=chunk_size
Aug 12 03:39:47 raspberrypi python[303]:   File "/usr/local/lib/python2.7/dist-p
Aug 12 03:39:47 raspberrypi python[303]:     stream = Stream(self, *args, **kwar
Aug 12 03:39:47 raspberrypi python[303]:   File "/usr/local/lib/python2.7/dist-p
Aug 12 03:39:47 raspberrypi python[303]:     self._stream = pa.open(**arguments)
Aug 12 03:39:47 raspberrypi python[303]: IOError: [Errno -9997] Invalid sample r
lines 1-18/18 (END)

EmerickH, 2019-01-07:
@tinkerermaruthu You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH

adumaraj, 2019-07-10:
When trying to clone to Raspberry Pi it is asking for Github userid and password. What are the values