alexa-pi/AlexaPi #415: CherryPy requires Python '>=3.5' but the running Python is 2.7.13

korre79, 2018-09-02:
Hi

I have a fresh install of rasbian stretch when i follow the instructions from this : https://github.com/alexa-pi/AlexaPi/wiki/Installation. 
I get this "CherryPy requires Python '>=3.5' but the running Python is 2.7.13" error and I can´t complete the installation. (scandisc 32Gb)
I did the the installation on a different sd-card (samsung 32Gb) but same RPI 3b+ with MagicMirror2 and i got everything to work.  I´m novice in this please help!
Peace!

DukeOfManville, 2018-09-02:
I am having this issue as well. I changed the default python to 3.5 as well as installed CherryPy using python 3 but to no avail.

neils84, 2018-09-04:
For the cherrypy issue, you need to manually install cherrypy thats greater than the error.

Type this:

cd
sudo pip install CherryPy==10.2.1

It will then install cherrypy without the python error.

You can then start the set up again.

isilher, 2018-10-07:
When I tried @neils84's tip, I was able to complete the installation but had trouble authorising the app via `localhost:5050`. I kept getting error 500 responses even though I double-tripple checked my config file. Eventually I found out the problem was with CherryPy. 

It turns out the installation script forces the use of pip2, which is based on Python 2. 

To solve it, I edited `/src/scripts/inc/common.sh` to call `pip3` in the `run_pip` method. Everything installed fine, although I had to manually run the authorisation script using `sudo python3 auth_web.py` to get through.

delta-alpha, 2018-10-11:
I just did sudo 

pip install CherryPy==10.1.1

The install script then passed this by as an already met dependency

henriz, 2018-10-15:
@isilher 
You can also channge python2 with python 3 in the run_python script. Then you don't have to run the auth_web.py manually.

thedavidhanks, 2018-10-26:
@henriz are you referring to the run_python fuction within https://github.com/alexa-pi/AlexaPi/blob/master/src/scripts/inc/common.sh ?