alexa-pi/AlexaPi #416: You haven't downloaded AlexaPi into /opt

JoelRybacki, 2018-09-06:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```Raspbian GNU/Linux 9
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 3 B+
```

**Python release (`python2 --version`):**
```
3.5
```

**Description of problem:**
```
When running setup.sh I am warned that I have not downloaded AlexaPi into /opt.  I have exactly followed the installation instructions and do see /opt/alexapi as the installation directory.  After ignoring the warning, I am able to continue the setup and confirmed that config.yaml contains the appropriate configuration.  However, alexapi.service is never started on reboot and I cannot manually start the service.  No log files exist and alexapi.service cannot be found.
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1.  
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

maso27, 2018-09-06:
Have you checked your capitalization scheme? 
  It should be under _/opt/AlexaPi_ not _/opt/alexapi_

If so, have you maybe got two AlexaPi directories nested?
  You should have a directory called _/opt/AlexaPi/src_ not _/opt/AlexaPi/AlexaPi/src_

JoelRybacki, 2018-09-11:
Don't have the device with me but I will give that a try later today.  The git created the directory structure and I know it was not in proper case but was definitely /opt/alexapi, not /opt/alexapi/alexapi.

EmerickH, 2019-01-07:
@JoelRybacki  You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH