alexa-pi/AlexaPi #422: HTTP Error 500 when going to http://raspberryIP:5050

JacobMeister, 2018-10-11:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian
```

**Your hardware platform and model you are running on:**
```
raspberry pi 3b+
```

**Python release (`python2 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
Can't reach localhost:5050 or raspberryIP:5050, the page is not working
```

**Expected:**
```
amazon login page
```

**Problem-relevant `config.yaml` entries:**
```
# Amazon Alexa settings
alexa:
  Client_ID: "amzn1.application-oa2-client.3f4eXXXXXXXXXXXXXXXXXXd7"
  Client_Secret: "da984c25XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX75127baa"
  Device_Type_ID: "SchoolProjectPi"
  Security_Profile_Description: "SchoolProjectPi"
  Security_Profile_ID: "amzn1.application.170b26XXXXXXXXXXXXXXXXXXXXXXXd20ef"
  refresh_token:

```

**Steps to reproduce:**

1. Follow instruction manual
2. manually install cherrypy because i get this error => https://github.com/alexa-pi/AlexaPi/issues/415
3. run setup normally
4. go to http://192.168.178.38:5050

**Traceback (if applicable):**
```

```

**Additional info:**
```
I had the cherrypy issue which i solved by installing it manually. The setup script ran without problems after that.
I've doublechecked every single ID and secret, so either i'm blind or something else is wrong.
```

[image]
[image]
[image]

skkuvind, 2018-10-12:
i am still getting this error "This page isn't working, localhost is currently unable to handle this request HTTP ERROR 500"

lucbors, 2018-10-12:
Take a look here: https://github.com/alexa-pi/AlexaPi/issues/420#issuecomment-429301592
It might work for you too. In my case the 500 is caused by an error when running the auth_web.py. After installing another library, it worked for me

JacobMeister, 2018-10-13:
Fixed it by following @lucbors instructions
It was indeed a missing version of the six library
Many thanks!

closed 2018-10-13 by JacobMeister