alexa-pi/AlexaPi #424: This method does not work on the current Raspbian version

henriz, 2018-10-15:
Please update it for Stretch, as this thing is broken on stretch.

If you run it, you'll run into dependency issues, as this thing uses pip2. Replacing pip2 with pip3, does finish the installation, but then the web_auth script doesn't work. Also replacing python2 with python3, gives garbled URLS in the authentication part, which don't work (lots of %20 and so forth in there, and in general the URLS are wrong (ie + "code", why?).

This method was being advertised as easier than using the official Alexa repo's, but it cost me about one and a half day of tinkering, and then still didn't work.

Expected result:
- Working Installation
God Result:
- Not working Installation

To reproduce, just get a current raspbian stretch image and follow the installation instructions. It's broken.

jdredd87, 2018-10-18:
It took me a few hours tonight but I finally got it working, using the latest Raspbian.

I to ran into issues with python / pip / blah

then having to use

sudo pyhton3 auth_web.py

to get the auth_web to work properly

then it was a matter of going through the Pulse Audio stuff and modify the config.yaml
to match up to that stuff and get the usb microphone in the input settings
and finally it works.

lorta, 2018-10-25:
Same here, it is broken due to python and pip versions, still no luck fixing it