alexa-pi/AlexaPi #426: !IMPORTANT!: Maintenance update

renekliment, 2018-10-31:
Hey there!

it seems many people are running into silly issues with this. Mainly because of dependencies and py2 / py3 - did Raspbian switch to py3 as default?

I've decided to do a little maintenance update so people who still wanna use this repo get less issues. I will try to do it on weekends when I get some free time. No promises about it though.

Until then, please don't open new issues regarding dependencies.

TODO list:
- [ ] use python 3 as default
- [ ] use _venv_ so we don't have clashing deps
- [ ] update the deps - they have been outdated for a while

mschindl, 2018-11-02:
I needed two workarounds to get it work.

https://github.com/alexa-pi/AlexaPi/issues/420#issuecomment-429301592
https://github.com/alexa-pi/AlexaPi/issues/415#issuecomment-418217508

renekliment, 2018-12-22:
Since there is probably nobody to test stuff, I've released this as `v1.7` and I hope for the best.

Also, what came to my mind only now is that C.H.I.P. computer is stuck with Jessie (and Python 3.4.x) and becomes unsupported with version. When installing a newer Python version manually, it should work and be okay.