alexa-pi/AlexaPi #430: Blank authentication page

Watta1, 2018-12-15:
**Your hardware platform and model you are running on:**
```
RaspberryPi 3 B+ (With google AIY latest image)
```

**Python release (`python2 --version`):**
```
Python 2.7.13

```

**Description of problem:**
```
Black page when the authentication page should load up.
Tried using the Ip of the machine on ther machines that are connected to the network with no luck.
Thanks a head. 
```

morridm, 2018-12-22:
I ran this to reinstall cherrypi and it fixed my blank auth page issue.  

**sudo pip3 install --upgrade --no-deps --force-reinstall cherrypy**

$ cd /opt/AlexaPi/src
/opt/AlexaPi/src $ sudo pip3 install --upgrade --no-deps --force-reinstall cherrypy
Collecting cherrypy
  Using cached https://files.pythonhosted.org/packages/3e/a2/cb5636c76781452bacbdb67d666b787ae560fb180d72eb5f94a891fcc03a/CherryPy-18.1.0-py2.py3-none-any.whl
Installing collected packages: cherrypy
  Found existing installation: CherryPy 18.1.0
    Uninstalling CherryPy-18.1.0:
      Successfully uninstalled CherryPy-18.1.0
Successfully installed cherrypy-18.1.0
pi@alejandro:/opt/AlexaPi/src $ sudo python3 auth_web.py
Ready goto http://192.168.1.93:5050 or http://localhost:5050  to begin the auth process
(Press Ctrl-C (or close this window on Windows) to exit this script once authorization is complete)

renekliment, 2018-12-22:
This should be fixed with the newly released `v1.7`. See #426.

closed 2018-12-22 by renekliment