alexa-pi/AlexaPi #440: Errors when launching for the first time using AIY Kit.

kunalbotla, 2019-01-13:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_
```
Latest version of Raspian Pixel (Version: November 2018, Release date: 2018-11-13, Kernel version: 4.14).
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 2B
```

**Python release (`python3 --version`):**
```
Python 3
```

**Description of problem:**
```
I am getting some errors when I am launching Alexa Pi. I am using the AIY kit so I am following [this](https://blog.skiilaa.me/running-alexa-on-aiy-voice-kit/) guide aswell as the offical setup guide. The response from the status is below.

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-01-13 02:36:00 GMT; 7min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 886 ExecStart=/usr/bin/python3 /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 886 (code=exited, status=1/FAILURE)

Jan 13 02:35:56 alexapi python3[886]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Jan 13 02:35:56 alexapi python3[886]: ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Jan 13 02:35:56 alexapi python3[886]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Jan 13 02:35:56 alexapi python3[886]: attempt to connect to server failed
Jan 13 02:35:58 alexapi python3[886]: INFO: Checking Internet Connection ...
Jan 13 02:35:58 alexapi python3[886]: INFO: Connection OK
Jan 13 02:35:58 alexapi python3[886]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installatio
Jan 13 02:36:00 alexapi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 02:36:00 alexapi systemd[1]: AlexaPi.service: Unit entered failed state.
Jan 13 02:36:00 alexapi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
```

**Expected:**
```
I expect it to run without any errors and detect my commands.
```

**Problem-relevant `config.yaml` entries:**
```
There is nothing in the refresh_token field of the config file.
```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

EmerickH, 2019-01-13:
Hello,
Please read https://github.com/alexa-pi/AlexaPi/wiki/Installation

kunalbotla, 2019-01-13:
@EmerickH I read the installation guide and followed it before. The issue started after setting it up.

kunalbotla, 2019-01-13:
I ran the setup script again and it worked.

closed 2019-01-13 by kunalbotla