alexa-pi/AlexaPi #443: NO GO On Ubuntu 17.10

Zibri, 2019-01-25:
I followed the guide.
First the setup script stopped when trying to install vlc-nox (which does not exist anymore)
I changed 
./AlexaPi/src/scripts/inc/os/debian.sh
from "vlc-nox" to "vlc"
The setup progressed a little more and then stopped again:
```
Removing jackd2-firewire (1.9.10+20150825git1ed50c92~dfsg-5ubuntu1) ...
dpkg: jackd2: dependency problems, but removing anyway as you requested:
 jackd depends on jackd2 | jackd1; however:
  Package jackd2 is to be removed.
  Package jackd1 is not installed.

Removing jackd2 (1.9.10+20150825git1ed50c92~dfsg-5ubuntu1) ...
Selecting previously unselected package jackd1.
(Reading database ... 329057 files and directories currently installed.)
Preparing to unpack .../jackd1_1%3a0.125.0-2_amd64.deb ...
Unpacking jackd1 (1:0.125.0-2) ...
dpkg: error processing archive /var/cache/apt/archives/jackd1_1%3a0.125.0-2_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libjackserver.so.0', which is also in package libjack-jackd2-0:amd64 1.9.10+20150825git1ed50c92~dfsg-5ubuntu1
Errors were encountered while processing:
 /var/cache/apt/archives/jackd1_1%3a0.125.0-2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

jleaders, 2019-02-12:
Can confirm this error in Ubuntu 18.10 as well E: `Package 'vlc-nox' has no installation candidate`

I also got
```
Found existing installation: PyYAML 3.12
Cannot uninstall 'PyYAML'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
``` and doing `sudo apt remove python-yaml` didn't help
But really, you should not be running pip on a deployment machine, rather you should just package your python project in a wheel or egg or something deployable.

EmerickH, 2019-04-14:
For vlc: https://github.com/alexa-pi/AlexaPi/issues/436
@jleaders You can also remove `pyyaml` from requirements.txt and install yourself. Remember that it's an unmaintained project...
@Zibri It's an apt problem not Alexapi one, try removing all jack packages and retry:
```
sudo apt remove jackd1 jackd2 libjack-jackd*
```

Alexvongtt, 2019-06-29:
It won't let me change the. "./AlexaPi/src/scripts/inc/os/debian.sh
from "vlc-nox" to "vlc"

Alexvongtt, 2019-06-29:
How did it change Debian.sh? Won’t let me save my edit

HemanthDavuluri, 2019-07-20:
yes it is not letting us change to vlc-nox to vlc-bin