alexa-pi/AlexaPi #449: after installation Alexa is not respawning and when i try to play or record anything doesnt work

Mohamed19966, 2019-02-19:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_
```4.14.79-v7+
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi model B
```

**Python release (`python3 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
i followed installation process and started Alexa but nothing happened so i tried to check my audio system 
when i tried "aplay /usr/share/sounds/alsa/Front_Center.wav"
"arecord -f S16_LE -r 16000 -D your_input_device test.wav"
"aplay -f S16_LE -r 16000 test.wav"
nothing happened even tho my audio system was working before installtion
and when i try to run it in debug mode 
i get "File "/opt/AlexaPi/src/main.py", line 18, in <module>
    import yaml
ImportError: No module named yaml
"
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Feb 19 03:26:53 raspberrypi python3[1638]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Feb 19 03:26:53 raspberrypi python3[1638]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Feb 19 03:26:53 raspberrypi python3[1638]: attempt to connect to server failed
Feb 19 03:26:54 raspberrypi python3[1638]: INFO: Checking Internet Connection ...
Feb 19 03:26:55 raspberrypi python3[1638]: INFO: Connection OK
Feb 19 03:26:55 raspberrypi python3[1638]: INFO: AVS token: Requesting a new one
Feb 19 03:26:56 raspberrypi python3[1638]: INFO: AVS token: Obtained successfully

```

Marvv90, 2019-04-14:
Hi, i have the same Issue, have you Solved the Problem ?

Mohamed19966, 2019-04-14:
No unfortunately

EmerickH, 2019-04-14:
Hello,
It seems you have multiples issues at the same time:
 * I think you have multiples versions of Python installed, and so when you run manually in debug mode you don't run the same version where AlexaPi installed PiYaml. Remove unused python versions or force version (`python3 /opt/AlexaPi/src/main.py -d`)
 * Your audio is not configured (or unknown PCM), take a look at: https://emerickh.github.io/AlexaPi-config/setup/alsa.html. If you still have a problem send me your config.yaml audio config and the output of `aplay -L` & `arecord -L`

Marvv90, 2019-04-15:
`
# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: DEBUG

sound:
  # Name of your microphone device: "raspberrypi"
  # leave empty for default
  input_device: "Siberia Raw Prism Headset: USB Audio (plughw:2,0)"

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "plughw:CARD=sndrpihifiberry,DEV=0"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0
`

`
#/etc/asound.conf
pcm.speakerbonnet {
   type hw card 1
}

pcm.dmixer {
   type dmix
   ipc_key 1024
   ipc_perm 0666
   slave {
     pcm "speakerbonnet"
     period_time 0
     period_size 1024
     buffer_size 8192
     rate 44100
     channels 2
   }
}

ctl.dmixer {
    type hw card 1
}

pcm.softvol {
    type softvol
    slave.pcm "dmixer"
    control.name "PCM"
    control.card 1
}

[... 370 lines cut ...]
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2019-04-15 12:01:50 DEBUG: Setting up playback handler: VlcHandler
2019-04-15 12:01:51 INFO: Checking Internet Connection ...
2019-04-15 12:01:51 DEBUG: Starting new HTTPS connection (1): api.amazon.com:443
2019-04-15 12:01:51 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2019-04-15 12:01:51 INFO: Connection OK
2019-04-15 12:01:51 INFO: AVS token: Requesting a new one
2019-04-15 12:01:51 DEBUG: Starting new HTTPS connection (1): api.amazon.com:443
2019-04-15 12:01:52 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 978
2019-04-15 12:01:52 INFO: AVS token: Obtained successfully
2019-04-15 12:01:52 DEBUG: Stopping audio play
2019-04-15 12:01:52 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2019-04-15 12:01:52 DEBUG: Player State: State.Opening
2019-04-15 12:01:52 DEBUG: Player State: State.Playing
2019-04-15 12:01:52 DEBUG: Started play. 
2019-04-15 12:01:55 DEBUG: Player State: State.Ended
2019-04-15 12:01:55 DEBUG: Finished play. 
`

No Sound played ... :(

Marvv90, 2019-04-15:
Thanks EmerickH ^^

EmerickH, 2019-04-16:
@Marvv90 you're welcome
Try:
```
su alexapi
python3 /opt/AlexaPi/src/main.py -d
```
(`exit` to come back to normal user after alexapi stop)

Marvv90, 2019-04-18:
Hi @EmerickH ,

I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!

marcusbaze, 2020-06-21:
> Hi @EmerickH ,
> 
> I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!

I am using a speaker bonnet on the GPIO pins of a raspberry pi zero how did you fix the issue?

KwokCT, 2020-08-18:
> Hi @EmerickH ,
> 
> I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!

Same using a speaker bonnet too. May I know how you fixed the issue please? TIA!

jacobkaufman, 2020-10-03:
> > Hi @EmerickH ,
> > I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!
> 
> Same using a speaker bonnet too. May I know how you fixed the issue please? TIA!

I fixed this issue by using the "dummy" platform instead of "raspberrypi".

Edit your config file `sudo nano /etc/opt/AlexaPi/config.yaml`

Go to the line `platform:` and replace `device: "raspberrypi"` with `device: "dummy"`

This will configure your device to stop using pins 18, 24 and 25 and you will lose the ability to use led's and a button press, and since you need 18 for the Adafruit Speaker Bonnet this should let the speakers work again.

I'm sure there is probably a way to just disable the use of pin 18 but since I am not going to be using the leds or a button for my project I didn't dive deeper into this.

After you can run a test by stopping the Alexa service and restarting it into debug mode. You should hear a "hello" come from your speaker in Alexa's voice if all goes well. Hopefully that helps!

Stop Alexa service `sudo systemctl stop AlexaPi.service`
Start Alexa service in debug mode `python3 /opt/AlexaPi/src/main.py -d`

jacobkaufman, 2020-10-03:
> > Hi @EmerickH ,
> > I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!
> 
> I am using a speaker bonnet on the GPIO pins of a raspberry pi zero how did you fix the issue?

https://github.com/alexa-pi/AlexaPi/issues/449#issuecomment-647151156

KwokCT, 2020-10-04:
> > > Hi @EmerickH ,
> > > I have solved my Issue, I use a I2S Soundcard & the declared GPIO Pins used the same as the Soundcard -.- ^^Very Tricky to found out ! But now all work's fine ! thanks for your help !!
> > 
> > 
> > Same using a speaker bonnet too. May I know how you fixed the issue please? TIA!
> 
> I fixed this issue by using the "dummy" platform instead of "raspberrypi".
> 
> Edit your config file `sudo nano /etc/opt/AlexaPi/config.yaml`
> 
> Go to the line `platform:` and replace `device: "raspberrypi"` with `device: "dummy"`
> 
> This will configure your device to stop using pins 18, 24 and 25 and you will lose the ability to use led's and a button press, and since you need 18 for the Adafruit Speaker Bonnet this should let the speakers work again.
> 
> I'm sure there is probably a way to just disable the use of pin 18 but since I am not going to be using the leds or a button for my project I didn't dive deeper into this.
> 
> After you can run a test by stopping the Alexa service and restarting it into debug mode. You should hear a "hello" come from your speaker in Alexa's voice if all goes well. Hopefully that helps!
> 
> Stop Alexa service `sudo systemctl stop AlexaPi.service`
> Start Alexa service in debug mode `python3 /opt/AlexaPi/src/main.py -d`
> 
> Edit: You can also change which pin you want to use for a button in the config file. For example I changed mine from 18 to 22 and now it works fine with the `raspberrypi` platform selected.

Thanks much for your help Jake! Much appreciated