alexa-pi/AlexaPi #451: AlexaPi - not picking up wakeword

SmakyTaky, 2019-03-27:
I have installed Raspbian Stretch on a Rasberry Pi 3+ with MagicMirror with the MMM-AlexaPi module installed and configured.

AlexaPi was installed to communicate with MagicMirror. I followed AlexaPi documentation to get it installed, I configured pulse audio to run in system-wide mode. 

Whenever I boot the system I get Magic Mirror to load properly followed by "hello" on the speakers signaling AlexaPi is working. Logfile for both MagicMirror & AlexaPi shows no communication errors.

However, I am unable to have AlexaPi to react to the keyword "Alexa" and something I noticed is that Alexa would from time to time "wake up" reacting to my voice randomly and even responding to me stating it could not understand me. 

[Debugging Script output](http://alsa-project.org/db/?f=f94dd597239e25825dc575e491aff5064d5025d4)

AlexaPi log...

```
mar 26 17:01:04 RasPiSMirror systemd[1]: Started Alexa client for all your devices.
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
mar 26 17:01:08 RasPiSMirror python3[26796]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
[... cut ...]
would yield a high pitch 1 sec or so sound.. no actual recording of anything I might have said that triggered pocketsphinx to identify a "wakeword".

Not sure where to go from here. As it seems AlexaPi is configured and managing sound properly (I can manually play and record sound using the above configuration - Speaker through jack & USB Microphone). Magic Mirror does connect to AlexaPi as there are not connection errors recorded but it seems AlexaPi is not either recording properly the sound or simply failing to recognize the wakeword.

SmakyTaky, 2019-03-27:
I just noticed the mp3 file is actually something that got recorded from what AlexaPi actually responded. Probably I should try some other paramters for aplay to play it properly. 

[b71506d672a260f13e757f00fa63fbc1.zip](https://github.com/alexa-pi/AlexaPi/files/3010997/b71506d672a260f13e757f00fa63fbc1.zip)

However, again the odd thing here is that AlexaPi is randomly picking sound while I am talking and responded... and seems to me it is even taking that as actual input for it.

SmakyTaky, 2019-04-06:
Anyone?

EmerickH, 2019-04-14:
Hello,
Your audio is not well configured (for audio output at least), please follow: https://emerickh.github.io/AlexaPi-config/setup/alsa.html

SmakyTaky, 2019-04-23:
> Your audio is not well configured (for audio output at least), please follow: https://emerickh.github.io/AlexaPi-config/setup/alsa.html

Well... not sure about that... I did modify the config.yaml file with:

input_device: "CMTECK MINI: USB Audio (plughw:1,0)"
output: "alsa"
output_device: "plughw:CARD=ALSA,DEV=0"

and I can clearly hear alexa saying "hello". It is the recording that seems not to be working properly,

SmakyTaky, 2019-04-23:
There are a few folders under tmp named  AlexaPi-runtime-_something_ but none of them have any files under them.

EmerickH, 2019-04-23:
Are you on a RPI Zero? They have the same issue here https://github.com/Kitt-AI/snowboy/issues/430
Try using PulseAudio:
https://emerickh.github.io/AlexaPi-config/setup/pulse.html
(you may have to [run it in system-wide mode](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended))