alexa-pi/AlexaPi #452: {Active: Failed} Tried to run debug but ImportError:No module named parse

ChanasitGithub, 2019-04-01:
**Your hardware platform and model you are running on:**
Hardware : RaspberryPI 3 model B rev 1.2
Python version : 3.5.3

**Description of problem:**
```
I install AlexPi on opt then check the status by run sudo systemctl status AlexaPi.service 
I got : ● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-04-01 07:26:14 EST; 1h 2min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 488 (code=exited, status=1/FAILURE)

Apr 01 07:26:09 raspberrypi python3[488]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Apr 01 07:26:09 raspberrypi python3[488]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Apr 01 07:26:09 raspberrypi python3[488]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Apr 01 07:26:09 raspberrypi python3[488]: attempt to connect to server failed
Apr 01 07:26:11 raspberrypi python3[488]: INFO: Checking Internet Connection ...
Apr 01 07:26:12 raspberrypi python3[488]: INFO: Connection OK
Apr 01 07:26:12 raspberrypi python3[488]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installation instructions).
Apr 01 07:26:14 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 07:26:14 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Apr 01 07:26:14 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

Then I tried to run dubug by : /opt/AlexaPi/src/main.py -d
I got : Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 23, in <module>
    import alexapi.tunein as tunein
  File "/opt/AlexaPi/src/alexapi/tunein.py", line 10, in <module>
    from urllib.parse import urlparse
ImportError: No module named parse
  
```
PS : Thank you .w.

SmakyTaky, 2019-04-06:
Same here... manually installed dependencies until I got to a point there are some that although installed it is still reported as missing

EmerickH, 2019-04-14:
Hello,
it seems you didn't run the install script/correctly followed the installation process.
Please follow: https://github.com/alexa-pi/AlexaPi/wiki/Installation

closed 2019-04-16 by ChanasitGithub

joshyeram, 2020-12-03:
did you guys fix it somehow? having the same error