alexa-pi/AlexaPi #465: vlc

HemanthDavuluri, 2019-07-20:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python3 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

Kleinstein-github, 2019-07-28:
Here is the FIX:

https://richardtech.net/2019/03/alexa-on-raspberry-pi/

renekliment, 2019-08-20:
It should be fixed now. If you could check with a clean install that would be awesome.

closed 2019-08-20 by renekliment