alexa-pi/AlexaPi #497: Alexa not responding after "yes"

hankp46, 2020-06-06:
**Your OS (including version) where you are running AlexaPi:**

Raspberry Pi OS_```
```

**Your hardware platform and model you are running on:**
```Raspberry Pi4

```

**Python release (`python3 --version`):**
```
Python 3.7.3
```

**Description of problem:**
```
I have Alexa installed  with a usb mic and audio out through the 3.5mm jack.
On start I hear "Hello" from Alexa and when I use the wake word I get the response "yes"
When I ask a question I get no response.
When I perform a Status i see the following error:
Jun 05 23:00:09 Hank python3[13791]: INFO: Triggered: pocketsphinx
Jun 05 23:00:13 Hank python3[13791]: INFO: (process_response Error) Status Code: 500

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```
arecord and aplay show audio input and output working
```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

jacobkaufman, 2020-10-17:
Have you tried to use snowboy instead of pocketsphix?

twilliams0805, 2021-11-12:
Sorry to resurrect an old question but noting this is still an open issue I assume you didn't find a solution to this error? Searched the internet but you appear to be the only one who has experienced the same error message.

DarrellH89, 2021-11-15:
You are not alone. I am getting the same error running on a Raspberry PI4

jacobkaufman, 2021-11-15:
> You are not alone. I am getting the same error running on a Raspberry PI4

> Sorry to resurrect an old question but noting this is still an open issue I assume you didn't find a solution to this error? Searched the internet but you appear to be the only one who has experienced the same error message as I'm getting.

Try using snowboy instead of pocketsphinx in the config file. This worked for me.

yulonglovecode, 2021-11-17:
> Try using snowboy instead of pocketsphinx in the config file. This worked for me.

How u changed snowboy instead of using pocketsphinx ?

ramandeep89, 2021-11-17:
Ok, I installed and enabled snowbow in the config.yaml but I am still getting the same Status Code 500 error and alexa just responses with a yes. Here are the debug logs on calling alexa
`
2021-11-17 21:10:53 INFO: Triggered: snowboy
2021-11-17 21:10:53 DEBUG: Stopping audio play
2021-11-17 21:10:53 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2021-11-17 21:10:53 DEBUG: Player State: State.Opening
2021-11-17 21:10:53 DEBUG: Player State: State.Playing
2021-11-17 21:10:53 DEBUG: Started play.
2021-11-17 21:10:54 DEBUG: Player State: State.Ended
2021-11-17 21:10:54 DEBUG: Finished play.
2021-11-17 21:10:54 DEBUG: Recording: Setting up
2021-11-17 21:10:54 DEBUG: Recording: Start
2021-11-17 21:10:54 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com:443
2021-11-17 21:10:55 DEBUG: Start sending speech to Alexa Voice Service
2021-11-17 21:10:56 DEBUG: Recording: End
2021-11-17 21:10:56 DEBUG: Finished sending speech to Alexa Voice Service
2021-11-17 21:11:00 DEBUG: Processing Request Response...
2021-11-17 21:11:00 INFO: (process_response Error) Status Code: 500
`

twilliams0805, 2021-11-17:
> Try using snowboy instead of pocketsphinx in the config file. This worked for me.

Unfortunately snowboy doesn't allow me to change to my own wakeword as the website has been closed down which is why I am using Pocketsphinx. I don't think this issue is related to Pockesphinx as there is no mention of it in the logs.

blackPantherOS, 2021-12-04:
This issues not related with pocketshinx, seems thies API chages issue because the AlexaPi uses old v1 API in https://access-alexa-na.amazon.com/v1/avs/speechrecognizer/recognize and the server response from this address is:
 '500 Internal seerver error'

https://developer.amazon.com/en-US/docs/alexa/alexa-voice-service/exceptions.html