alexa-pi/AlexaPi #500: Alexa dont wake up

uok825, 2020-06-17:
I hear Hello! but alexa dont wake up when i say Alexa!. Help me please.

ericmil87, 2020-07-12:
system status looks ok.. but I alexa wont say anything... or wake with alexa word

londer1995, 2020-07-27:
If you get 0 errors (also with debug), try to say "alexaalexaalexaalexa", more than once. 
I use an external speaker (jabra 510) and seems like there are some "blackout" moments when the device starts hearing. From the debug, I can see alexaPi is hearing but still she can't get the first/second time.
If anyone can help resolve this issue, could be usefull.

p.s. please share more info when asking for help..

jacobkaufman, 2020-10-17:
I was having this issue too, then I switched to use 'snowboy' and set the threshold to '1', which is the max. For some reason 'pocketsphix' wasn't working for me.

All this can be configured in your 'config.yaml' file.

ScratpythHTML, 2020-12-18:
I do not have a config.yaml file at all. Has anybody else had this issue?
I have the config.template.yaml file but I cannot edit it in any way.

Anil-v29, 2021-06-26:
Same issue on Pi2, Can somebody help me to fix this!