alexa-pi/AlexaPi #504: something went wrong  pls help

hidebook, 2020-09-12:
hello i was using a tutorial to install AlexaPi and when i entered 
`sudo systemctl status AlexaPi.service` it came up with
```
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: **_failed_** (Result: exit-code) since Sat 2020-09-12 20:57:16 CEST; 15s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 7783 ExecStart=/usr/bin/python3 /opt/AlexaPi/src/main.py --daemon **_(code=exited, status=1/FAILURE)_**
 Main PID: 7783 (code=exited, status=1/FAILURE)

Sep 12 20:57:12 raspberrypi python3[7783]: Cannot connect to server request channel
Sep 12 20:57:12 raspberrypi python3[7783]: jack server is not running or cannot be started
Sep 12 20:57:12 raspberrypi python3[7783]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Sep 12 20:57:13 raspberrypi python3[7783]: /opt/AlexaPi/src/main.py:71: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.o
Sep 12 20:57:13 raspberrypi python3[7783]:   config = yaml.load(stream)
Sep 12 20:57:13 raspberrypi python3[7783]: INFO: Checking Internet Connection ...
Sep 12 20:57:14 raspberrypi python3[7783]: INFO: Connection OK
Sep 12 20:57:14 raspberrypi python3[7783]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installation instructions).
Sep 12 20:57:16 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Sep 12 20:57:16 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
~

```
```
`` because i'm not an expert could someone please help me?`
`

hidebook, 2020-09-13:
and when I open the page https://localhost:5050 or http://localhost:5050 there is an Amazon error page and it says:
```
Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
Request Details
client_id=amzn1.application-oa2-client.4fea7cfc27074f09ae2a9cbc24942019
scope=alexa%3Aall
scope_data=%7B%22alexa%3Aall%22%3A+%7B%22productID%22%3A+%22AlexaPi%22%2C+%22productInstanceAttributes%22%3A+%7B%22deviceSerialNumber%22%3A+%223392a7ef74e3085c6f910e00820f307dcef3e16a3851c95d7aa7522aae14bb6e%22%7D%7D%7D
response_type=code
redirect_uri=http%3A%2F%2Flocalhost%3A5050%2Fcode
```
What should I do?

LonelyEagle29, 2020-12-22:
Are you trying to connect from another computer, or in the same machine that AlexaPi is installed? Did u open the page before closing the script? You have to access the address "HTTP://YOUR.IP.ADD.RESS:5050 **before** closing the script. You only close the script in the terminal after the page in your browser tells you to do so.