alexa-pi/AlexaPi #506: Adding digitalWrite to control LED animation breaks the app

Oversixty, 2020-11-02:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_
```
Ver 10 Buster all updates and upgrades as of this date. 
```

**Your hardware platform and model you are running on:**
```
RPI 4B
AIY Tophat audio board. 
```

**Python release (`python3 --version`):** 
```
2.7.16
```

**Description of problem:**
```
After following the instructions to make the LED light up I get errors when compiling. https://developer.amazon.com/en-US/docs/alexa/alexa-voice-service/indicate-device-state-with-leds.html
At this point I have tried a lot of different configurations and have found with all of the code changes per the instructions except adding the digitalWrite code the Alexa app works fine. However as soon as I add 1 digitalWrite command I get errors. 
I downloaded the latest WiringPi libraries for RPI 4 (ver 2.52) and can run the sample Blinky code with no problems. 
```

**Expected:**
```
To turn on a light. 
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. use digitalWrite in a script

 case DialogUXState::IDLE:
                digitalWrite (0, LOW);
                ConsolePrinter::prettyPrint("Alexa is currently idle!");
                return; 

**Traceback (if applicable):**
```

```

**Additional info:**
```
/usr/bin/ld: ../../../SampleApp/src/libLibSampleApp.so: undefined reference to `digitalWrite'
collect2: error: ld returned 1 exit status
make[3]: *** [applications/acsdkPreviewAlexaClient/src/CMakeFiles/PreviewAlexaClient.dir/build.make:149: applications/acsdkPreviewAlexaClient/src/PreviewAlexaClient] Error 1
make[2]: *** [CMakeFiles/Makefile2:4473: applications/acsdkPreviewAlexaClient/src/CMakeFiles/PreviewAlexaClient.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:4485: applications/acsdkPreviewAlexaClient/src/CMakeFiles/PreviewAlexaClient.dir/rule] Error 2
make: *** [Makefile:1163: PreviewAlexaClient] Error 2

```

lilyt86, 2020-11-09:
excuse me if I comment without an answer. I also have the same problem and I would like to read the answer they will give you in the future

renekliment, 2020-11-10:
I think you are in a wrong repository. This is not the official Amazon stuff - https://github.com/alexa/avs-device-sdk