alexa-pi/AlexaPi #508: Help me to add a “Neopixel WS2812B device state indicator” for Alexa Pi

iamsujay, 2020-11-21:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 3 Model B Rev 1.2

```

**Python release (`python3 --version`):**
```
Python 3.7.3

```

**Description of problem:**
```
I can't see any appropriate "indicator" file to edit
```

**Expected:**
```
A device state indicator with Neopixel
```

**Problem-relevant `config.yaml` entries:**
```
NO
```

**Steps to reproduce:**

1.  NA
2.  NA
3.  NA

**Traceback (if applicable):**
```
NA
```

**Additional info:**
```
I would like to add a 5volt “Neopixel WS2812B” led strip as a device state indicator for Alexa Pi

I don't need any fancy animation. I just want

Alexa on idle> Neopixel off.
Wake word Alexa> Neopixel on.
Alexa thinking> Simple animation.
Alexa speaks> Simple animations.
```

renekliment, 2020-11-21:
You can completely customize the indicators in here: https://github.com/alexa-pi/AlexaPi/tree/master/src/alexapi/device_platforms
You can either edit one that suits you best or better make your own using an existing platform file as a template.

However I don't recommend sticking with this project as it has been abandoned for quite a while - use at your own _risk_. ;-)

iamsujay, 2020-11-22:
Let me check that 😊

closed 2020-11-22 by iamsujay