alexa-pi/AlexaPi #511: Config file non-existent

ScratpythHTML, 2020-12-18:
I am using a raspberry pi 0W

When I run the program using `sudo systemctl start AlexaPi.service` and check the status by running `sudo systemctl status AlexaPi.service` I have a few messages:

Cannot connevct to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr : :~JackShmReadWritePtr - Init not done for -1, skipping unlock
/opt/AlexaPi/src/main.py:71: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details
    config = yaml.load(stream)
INFO: Checking Internet Connection ...
INFO: Connection OK
INFO: AVS token: Requesting a new one
INFO AVS token: Obtained successfully

I have cloned the repo into /opt/. There is no config file in the first AlexaPi folder, in the AlexaPi/src folder there is a file called config.template.yaml. However, I can not edit this file to change the audio devices or the wake-word. If I try to edit this file in any way it tells me that I do not have permission. This file does not have any of my personal AVS security details even though I entered them during the setup.

It would be great if someone knew how to fix this issue as I need to change the settings of the config file but it is non-existent. The fact that it can't connect to the server is also worrying.

renekliment, 2020-12-19:
https://github.com/alexa-pi/AlexaPi/wiki/Configuration-file

closed 2020-12-19 by renekliment

ScratpythHTML, 2020-12-19:
Thank you for directing me to this document. However there is no config.yaml file on my pi at all. It would be great if you could explain how to get one.

renekliment, 2020-12-19:
If it's not there for some weird reason, just create it with the contents of the template file. However, you need to fill out the auth info. I think better option is running the install again, watching out if any erorrs happen during the installation that should create the config and fill out all the important things.

ScratpythHTML, 2020-12-19:
I'll try that. Thank you so much!

ScratpythHTML, 2020-12-20:
**I ran the install again and now have two issues:**
I now have a config.yaml file in `/etc/opt/AlexaPi` and the rest of the program in `/opt/AlexaPi`. However, I do not have any way to edit the config file in `/etc/opt/AlexaPi` as it say's that I do not have permission.

At the end of the install I received four errors:
`error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.`
`fatal: the remote end hugh up unexpectedly`
`fatal: early E0F`
`fatal: index-pack failed`
I don't know what any of these things mean but I can infer that they aren't good.

I also tested my audio and there is no problem there as far as I'm concerned.