alexa-pi/AlexaPi #513: Language cannot setup

ful4n1t0c0sme, 2020-12-19:
Hi. I read in the wiki:
Can I switch the language to German, etc?

Yes, just make sure your Account is a German one. Anyways you will pronounce "Alexa" in English but she will answer in German and recognize German too.

But my account is in es-MX and speasks me in english. I can not see any way to change the language. The wiki explanation is very vague.

I have seen that many who ask the same thing are sent to read the wiki at this point. But it doesn't explain anything.

How can I change my language settings then?

LonelyEagle29, 2020-12-22:
I speak Brazilian Portuguese, and since I connected my account, my alexa works in Portuguese (she answers "yes" in English, but after that understands PT-BR). The development is inactive (really sad, I would like to use alarm and timers) and the devs can't solve this, probably is some configuration on _your_ amazon account.

ful4n1t0c0sme, 2020-12-23:
Hi. My account is in es-MX and can not see any option in the Amazon account
to change to es-ES to try at least if then can speak in spanish to me.
I try creating a new account (es-ES) and it's the same. Only english talk.
It's sad because i do not speak english.