alexa-pi/AlexaPi #515: Alexa cannot set Alarms

prajul-L850, 2021-01-12:
Hello,I'm running alexa on a laptop with elementary os. It's been a day since I've installed AlexaPi on my system. Today morning,I tried to set an alarm(for the first time after the installation) but it couldn't do anything. After accepting the time for the alarm, it always says like this:
"Sorry,there as a problem setting your alarm. Please reconnect your device.

LonelyEagle29, 2021-01-14:
Please search through the issues before posting. This is already known, and probably will never be supported, since AlexaPi uses the old Amazon API, that doesn't support alarms, timers, and reminders, and the development is stopped.
Please see #446, #50 and #398.

prajul-L850, 2021-01-17:
OK.Thank you for letting me know

closed 2021-01-17 by prajul-L850