alexa-pi/AlexaPi #517: (process_response Error) Status Code: 500

yulonglovecode, 2021-11-17:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues]() before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
_Note: Raspbian older than Stretch is not supported!_
```
```

**Your hardware platform and model you are running on:**

raspberry pi 4b

**Python release (`python3 --version`):**

Python 3.7.3

**Description of problem:**
no response after speaking " hi" 

sudo systemctl status AlexaPi.service
shows this 
Nov 17 14:37:21 raspberrypi python3[493]: INFO: Triggered: pocketsphinx
Nov 17 14:37:30 raspberrypi python3[493]: INFO: (process_response Error) Status Code: 500

**Expected:**
to response after "key word detection"

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

ful4n1t0c0sme, 2021-11-20:
Same here. I think Amazon AVS V1 does not work anymore.
Problem is AVS V2 works (respeaker/avs), but no smart home support at all.

supercrab, 2021-11-23:
I receive the same issue too!  Gutted as I told my son I'd make him a talking skull. :(

sammachin, 2022-01-02:
Yep looks like amazon have turned off the v1 API around the end of October.

I'm working on a brand new project that uses the v2 API within Node-RED should have something to release in the next day or so, that should allow a lot more customisation and working with other hardware too.

closed 2022-01-02 by sammachin