alexa-pi/AlexaPi #520: After Saying hello it is not working with error status code 500

SahuPrabhat, 2022-01-15:
● AlexaPi.service - Alexa client for all your devices
     Loaded: loaded (/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-01-15 13:35:42 GMT; 52s ago
       Docs: https://github.com/alexa-pi/AlexaPi/wiki
   Main PID: 2647 (python3)
      Tasks: 4 (limit: 1597)
        CPU: 6.476s
     CGroup: /system.slice/AlexaPi.service
             └─2647 /usr/bin/python3 /opt/AlexaPi/src/main.py --daemon

Jan 15 13:35:44 raspberrypi python3[2647]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Jan 15 13:35:44 raspberrypi python3[2647]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Jan 15 13:35:44 raspberrypi python3[2647]: connect(2) call to /dev/shm/jack-998/default/jack_0 failed (err=No such file or directory)
Jan 15 13:35:44 raspberrypi python3[2647]: attempt to connect to server failed
Jan 15 13:35:45 raspberrypi python3[2647]: INFO: Checking Internet Connection ...
Jan 15 13:35:47 raspberrypi python3[2647]: INFO: Connection OK
Jan 15 13:35:47 raspberrypi python3[2647]: INFO: AVS token: Requesting a new one
Jan 15 13:35:48 raspberrypi python3[2647]: INFO: AVS token: Obtained successfully
Jan 15 13:36:22 raspberrypi python3[2647]: INFO: Triggered: pocketsphinx
Jan 15 13:36:30 raspberrypi python3[2647]: INFO: (process_response Error) Status Code: 500

blackPantherOS, 2022-01-16:
look these topics:
https://github.com/alexa-pi/AlexaPi/issues/497
https://github.com/alexa-pi/AlexaPi/issues/517

SahuPrabhat, 2022-01-16:
> look these topics: #497 #517

Thank you @blackPantherOS for the response but the case #497 and #517 are still open. Probably no solution is there.

tucano3000, 2022-01-28:
I tried yesterday and today and I'm also having the same problem. AlexaPi.Service status returns (process_response Error) Status Code:500 every time I ask a question.

dmzoneill, 2022-05-21:
This project seems dead :(

I updated the code to include output of the 500 response
```
        if response.status_code != 200:
                logger.info("(process_response Error) Status Code: %s", response.status_code)
                logger.info(response.content)
                response.connection.close()
                platform.indicate_failure()
                return

```

after runing in debug mode.
/usr/bin/python3 src/main.py --debug
```
(process_response Error) Status Code: 500
2022-05-21 22:26:37 INFO: b'<html>\n<head>\n<meta http-equiv="Content-Type" content="text/html; charset=utf-8" />\n<title>We\'re sorry!</title>\n</head>\n\n<body style="padding: 4em; color: black; background-color: white; font-family: Verdana,Arial,Helvetica,sans-serif;">\n\n<h2>We\'re sorry!</h2>\n<p>\nAn error occurred when we tried to process your request.\nRest assured, we\'re already working on the problem and expect to resolve it shortly.\n</p>\n\n<h2>D\xc3\xa9sol\xc3\xa9s!</h2>\n<p>\nUne erreur s\'est produite lorsque nous avons tent\xc3\xa9 de traiter votre requ\xc3\xaate.\nSoyez assur\xc3\xa9 que nous travaillons d\xc3\xa9j\xc3\xa0 \xc3\xa0 la r\xc3\xa9solution du probl\xc3\xa8me que nous pensons trouver tr\xc3\xa8s rapidement.\n</p>\n\n</body>\n</html>\n'
```
I've tried every url imagineable :/
[image]

officialasit, 2022-10-13:
Any update to this issue? Even Im getting a 504 Error after triggering
[image]

T-tronicks, 2022-11-13:
Same problem here.
Getting Code 504.

Amazon's REST API says:
504 | DEVICE_NOT_REACHABLE | Device is offline or unreachable.

So it seems amazon can't find my PI or sees it as offline.

pi@Brummel:~ $ sudo systemctl status AlexaPi.service
? AlexaPi.service - Alexa client for all your devices
     Loaded: loaded (/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-11-13 23:36:43 CET; 9min ago
       Docs: https://github.com/alexa-pi/AlexaPi/wiki
   Main PID: 2049 (python3)
      Tasks: 4 (limit: 415)
        CPU: 6min 40.314s
     CGroup: /system.slice/AlexaPi.service
             ??2049 /usr/bin/python3 /opt/AlexaPi/src/main.py --daemon

Nov 13 23:37:03 Brummel python3[2049]: INFO: Connection OK
Nov 13 23:37:03 Brummel python3[2049]: INFO: AVS token: Requesting a new one
Nov 13 23:37:05 Brummel python3[2049]: INFO: AVS token: Obtained successfully
Nov 13 23:37:08 Brummel python3[2049]: [02d7c978] alsa audio output error: cannot estimate delay: Broken pipe
Nov 13 23:37:52 Brummel python3[2049]: INFO: Triggered: platform
Nov 13 23:38:05 Brummel python3[2049]: INFO: (process_response Error) Status Code: 504
Nov 13 23:45:59 Brummel python3[2049]: INFO: Triggered: platform
Nov 13 23:46:12 Brummel python3[2049]: INFO: (process_response Error) Status Code: 504
Nov 13 23:46:15 Brummel python3[2049]: INFO: Triggered: platform
Nov 13 23:46:24 Brummel python3[2049]: INFO: (process_response Error) Status Code: 504

Anyone any luck

drservomonkey, 2023-01-12:
hi
even i am getting the 504 error and am not able to get a response beyond "yes" and "hello"
this is how the debug mode logs are:
2023-01-12 16:21:19 INFO: Triggered: pocketsphinx
2023-01-12 16:21:19 DEBUG: Stopping audio play
2023-01-12 16:21:19 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2023-01-12 16:21:19 DEBUG: Player State: State.Opening
2023-01-12 16:21:19 DEBUG: Player State: State.Playing
2023-01-12 16:21:19 DEBUG: Started play. 
2023-01-12 16:21:20 DEBUG: Player State: State.Ended
2023-01-12 16:21:20 DEBUG: Finished play. 
2023-01-12 16:21:20 DEBUG: Recording: Setting up
2023-01-12 16:21:20 DEBUG: Recording: Start
2023-01-12 16:21:20 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com:443
2023-01-12 16:21:54 DEBUG: Start sending speech to Alexa Voice Service
2023-01-12 16:21:55 DEBUG: Recording: End
2023-01-12 16:21:55 DEBUG: Finished sending speech to Alexa Voice Service
2023-01-12 16:22:00 DEBUG: Processing Request Response...
2023-01-12 16:22:00 INFO: (process_response Error) Status Code: 504

i am a noob so i might not be able to comprehend too much jargon \:
thanks in advance