alexa-pi/AlexaPi #524: Don´t can acess the authorization URL

Mgm0lm, 2023-02-11:
I don't know if the problem is on my raspberry pi, but after installation I can't access http://192.168.2.124:5050 or http://localhost:5050

drservomonkey, 2023-02-21:
in your amazon developer portal set allowed return urls as http://my.pi.ip:5050  and allowed origins as http://my.pi.ip:5050/code
you can replace my.pi.ip to your pi's ip or localhost
i faced the same problem but it worked when i opened this url from my pi's browser

alexbue01, 2023-10-09:
Hi All,
I have the same issue.
Have now installed a 2nd time a new Raspi.
Raspi image via raspberyy Pi Imager -> Raspberry Pi OS (32-bit)

Run all steps as desicribet in the manual
`pi@raspberrypi:/opt $ sudo systemctl start AlexaPi.service
pi@raspberrypi:/opt $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
     Loaded: loaded (/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2023-10-09 14:19:15 CEST; 646ms ago
       Docs: https://github.com/alexa-pi/AlexaPi/wiki
    Process: 5223 ExecStart=/usr/bin/python3 /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
   Main PID: 5223 (code=exited, status=1/FAILURE)
        CPU: 3.494s

Okt 09 14:19:11 raspberrypi python3[5223]: ALSA lib conf.c:5200:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
Okt 09 14:19:11 raspberrypi python3[5223]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
Okt 09 14:19:11 raspberrypi python3[5223]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Okt 09 14:19:11 raspberrypi python3[5223]: ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Okt 09 14:19:12 raspberrypi python3[5223]: INFO: Checking Internet Connection ...
Okt 09 14:19:13 raspberrypi python3[5223]: INFO: Connection OK
Okt 09 14:19:13 raspberrypi python3[5223]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project w>
Okt 09 14:19:15 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Okt 09 14:19:15 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
Okt 09 14:19:15 raspberrypi systemd[1]: AlexaPi.service: Consumed 3.494s CPU time.
lines 1-18/18 (END)`

Ip or Host doest work as well from the Raspi desktop